corygrant/dingoPDM #1: How can we obtain pre-assembled units?

ElDominio, 2023-02-09:
I know a large group of software guys who might be interested in acquiring a pre-assembled unit to play around with code. Where can I point them to? Is there a Slack channel for this project?

corygrant, 2023-02-09:
Currently the only units are the ones I've built, but if you'd like we can talk about building some more. 
There's no Slack channel yet, but I could create one if you like?

bryanbendall, 2023-02-16:
I would be interested in it as well.

corygrant, 2023-02-16:
I'm going to build a batch of V4 boards soon, would you be interested in buying an assembled unit?

bryanbendall, 2023-02-16:
Yes an assembled one would be good. I would like to try it out with the software I'm working on. I made some of my own boards but they need to be updated. Hard to find parts right now. Any idea on the cost of them yet?

corygrant, 2023-02-17:
It's probably not cost effective (or timely) for me to do assembly by hand. 
Purchasing parts from Digikey/Mouser the BOM cost is around $150 right now, plus maybe $4 for a PCB. 
But, it takes me hours to assemble one of these by hand. 

Maybe the next step should be getting this setup for PCBA through JLC or similar.

JZXPacky, 2023-06-27:
Any chance you've got this setup for JLC etc yet? I'm very interested in building one of these units and incorporating into my car with a Haltech Elite 2500.

corygrant, 2023-06-29:
@JZXPacky no not yet. I've done a little testing when I've been able to find the time, but not to a point that I would want to order more assembled boards. 
Feel free to use the Kicad files and BOMs to set it up though!

corygrant, 2024-04-24:
Just finished a batch of pre built PDMs. 
Working on getting the online shop set up, but PayPal also works

ZHoob2004, 2024-05-06:
Based on the new pictures in the repo, it looks like you have some units in hand. I could be interested in buying one, but I'm not sure how to contact you outside of github issues.

corygrant, 2024-05-06:
@ZHoob2004 I just opened the store site up, first shipments from the store going out Monday 5/13. 
https://dingo-electronics.square.site/

closed 2024-05-06 by corygrant