corygrant/dingoPDM #5: Is there software for this yet?

MatteBSki, 2023-06-01:
I came across this and its something I'd love to help develop.  I've been working on creating a multi PDM system that uses smaller PDMs around the vehicle to keep harness size and current draw to a minimum.  While yours is a single setup, I dont see why the concept couldnt be modified for different use cases.  

What I didnt see is any type of control software.  Is this current being programmed straight from code or is there something like TunerStudio being used as a GUI for flashing and monitoring the module?

corygrant, 2023-06-02:
The firmware and software are both in progress, but I've been super busy at work this year so I haven't been able to get them done as fast as I would like. 
The firmware is currently set up to be built with the settings hard-coded in, that's what I've been using for testing. 
But lately I've been working on the program to get status/send new settings from a PC over CAN or USB. 

Having no background in this type of thing I spent way too long trying to figure out how to make settings changes over CAN and USB work. 
Over USB is pretty easy as you can stuff a lot of data into a few messages, but CAN made it a lot more restrictive. 
In the end I decided to use an alpha numeric upper/lower case prefix system with identical data sets between CAN and USB. 
That seems to work pretty well, but it needs more development and testing. 

I'm hoping to have it finished by the end of the year. 

As far as the size of the unit, I went back and forth between different output counts and form factors for a while before landing on the 12 outputs. 
There are a lot of units out there that have many more, but for my use cases 12 outputs is plenty. 
If I need more, it's easy enough to use 2 units as the size is still fairly compact. 
I could definitely make it more compact if I left some hardware features out, but at such low volumes it's much easier to leave them on for possible future developments. 

Are you interested in helping with the firmware/software?

MatteBSki, 2023-06-02:
The amount of outputs is fine, no issue at all.  I've been also slowly designing a system that would have micro PDMs instead of a single one for ease of install.  Basicallly, you'd have a module for front lighting, horn, fans.  In the back, youd have one that handles the running lights, turns, brakes, fuel pump, then one for running loads under the dash along with a CAN input board for switches and such.  The thinking is that you can wire the car with harnesses that are only a few feet long and then interconnect them with a bat lead and CAN bus.  

I'd definitely be happy to help, I'd need to build a board first, but you have some pretty interesting designs and if I can help get them working, I'd be down.

corygrant, 2023-06-02:
That's definitely a cool way to do it, it seems like Link is trying to go that way with their Razor PDM. 

I decided against a more distributed approach due to:
-Weatherproofing and temperature requirements for under hood (I use these in rally cars, so extreme weather resistance required)
-Needing to run large fused conductors from the battery to each device, rather than 1 large fused conductor that I can branch between multiple devices locally
-Needing to have different spares with different settings pre-loaded for fast changes in service/pits (important for race/rally situations)

MatteBSki, 2023-06-02:
Yeah, waterproofing and building to be reliable at 125c for extended periods has been a hangup of mine as I've been choosing components.  Good thing is that when you run less loads per module, the heating from them is less as well so you can work more on heatsinking.  3D printing an enclosure may not work, but there are sealed enclosures available that will work. Similarly, since each module may only be responsible for 25-30A in general, you can power it with a 12-10AWG lead instead of a 4 or 6AWG lead.  

Regarding having to have separately programmed modules, that I would differ.  I want to use a single universal module and be able to program them over CAN (preferablly) from a central location, so as long as each has module has a bootloader, you should be able to plug in another and flash it in the car with a laptop.

I'd think that if the base programming of the Dingo is built out correctly, you should be able to do all that with your design using multiple modules versus a mini version of the same system (just gotta keep an eye on modularity in code).

rusefillc, 2024-01-30:
Just checking how SW is going :) is there a URL for anything-firmware or is it not publicly available in any shape or form?

rusefillc, 2024-01-30:
PS: there are screen shots at https://corygrant.github.io/DingoPDM/software/introduction/ so something seems to exist :)

corygrant, 2024-01-31:
Yes something does exist, good find! :)
I was making good progress on it, but then things got busy with the holidays and work projects. 
Hopefully finish it up soon and make it public. 
I did get 10 assembled boards in along with heatsink plates. 
Just need to finish printing some more cases.

danamaniac, 2024-05-05:
Hi mate thankyou for your devalopment on this, i have 5 boards made. Just waiting on the software. Hows it coming along?

corygrant, 2024-05-05:
Went ahead and made it public.
Still working on it, but it has enough functionality to use. 
https://github.com/corygrant/DingoConfigurator

And the firmware 
https://github.com/corygrant/DingoPDM_FW

closed 2024-05-05 by corygrant

danamaniac, 2024-05-06:
Thankyou Thankyou, legend.

https://github.com/corygrant/DingoConfigurator

https://github.com/corygrant/DingoPDM_FW