corygrant/dingoPDM #8: Holes Misaligned Between Baseplate and Case

ZHoob2004, 2024-04-23:
First of all, I really like the project, and I'm looking into having a few units produced so I can start to play with them and help out.

While going through the release files you've provided, I noticed that 3 of the 4 holes do not line up between the case and the base plate. Assuming this was meant to be a rectangular pattern, I determined the error is in the case and I'm working on a fix.

Related to this issue, is there any reason behind the chosen bolt pattern? I measure the bolt square as 96.39x87.25mm, which both seem like pretty random numbers. I feel like it would be beneficial to use a nice square number so it's easier to mark and drill holes for installation.

corygrant, 2024-04-24:
Weird! I have made a bunch of cases but haven't noticed any misalignment, I can check the source model later. 

Regarding the bolt pattern, I think that may be aligned to inches in Kicad but I'm not certain. I probably scooted the bolt holes around to improve trace placement to make life easier for myself haha. 

I like use a paper template with an optical punch for stuff like this so the round numbers aren't a big deal. 

By the way, I just finished a batch of PDMs. 
Working on getting the online shop up, but PayPal works too if you'd rather start with a pre-built one.

corygrant, 2024-04-24:
Good catch @ZHoob2004! 
I had moved the case up to make room for the thermal pads but accidentaly bumped it over as well. 
That explains why one screw was always a bit harder to get in and why the USB hole was a bit misaligned. 

Case step file is updated now.

I also see what you're saying about the holes being at weird dimensions. 
The hole positions came from a very early sketch where I had placed them based on the edge of the board. 
[image]

I'll look at fixing that spacing on the next batch of cases/baseplates

ZHoob2004, 2024-04-24:
Sounds good.

Another thing I ran into is the baseplate and heatsink are step files, but they probably only need to be dxf since they're two dimensional.

Normally this would be a non issue but I was quoting the parts for laser cutting and the baseplate was problematic with the modeled threads (which is what led to me finding the misalignment)

And I could be interested in a prebuilt unit if you have one. I can always reprint the case and change the baseplate if needed to change the bolt pattern.

closed 2024-05-21 by corygrant